PrintMakerLab/mks-wifi-plugin #249: Plugin can't be loaded.

WarlockSVG, 2021-12-02:
### Plugin Version

1.2.6

### Cura Version

4.12.1

### Platform

Windows 10

### Printer

FLSUN QQ-S Pro

### Reproduction steps

Copied the plugin to the plugin folder and started Cura.

### Actual results

[image]

I can't use the plugin... I get the error msg shown on the provided image.

These are the error msg from the log file

2021-12-02 13:57:20,915 - ERROR - [MainThread] UM.Logger.logException [112]: Exception: Error loading plugin mks-wifi-plugin-develop:
2021-12-02 13:57:20,918 - ERROR - [MainThread] UM.Logger.logException [116]: Traceback (most recent call last):
2021-12-02 13:57:20,919 - ERROR - [MainThread] UM.Logger.logException [116]:   File "C:\build\Cura\4.12-exe\build\inst\lib\python3.8\site-packages\UM\PluginRegistry.py", line 444, in loadPlugin
2021-12-02 13:57:20,920 - ERROR - [MainThread] UM.Logger.logException [116]:   File "C:\Users\Warlock\AppData\Roaming\cura\4.12\plugins\mks-wifi-plugin-develop\mks-wifi-plugin-develop\__init__.py", line 23, in register
2021-12-02 13:57:20,922 - ERROR - [MainThread] UM.Logger.logException [116]:     "output_device": MKSOutputDevicePlugin.MKSOutputDevicePlugin(),
2021-12-02 13:57:20,924 - ERROR - [MainThread] UM.Logger.logException [116]:   File "C:\Users\Warlock\AppData\Roaming\cura\4.12\plugins\mks-wifi-plugin-develop\mks-wifi-plugin-develop\MKSOutputDevicePlugin.py", line 45, in __init__
2021-12-02 13:57:20,926 - ERROR - [MainThread] UM.Logger.logException [116]:     Logger.log("d", self._preferences.getValue(Constants.MANUAL_INSTANCES).split(","))
2021-12-02 13:57:20,927 - ERROR - [MainThread] UM.Logger.logException [116]: AttributeError: 'NoneType' object has no attribute 'split'

### Expected results

It should work.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/7642183/cura.log
[image]
)

Elkin-Vasily, 2021-12-02:
Could you please manually delete the plugin directory and install it from Cura Marketplace? It calls MKS WIFI Plugin there.

Elkin-Vasily, 2021-12-08:
@WarlockSVG any updates?

Elkin-Vasily, 2021-12-25:
Closed due to inactivity

closed 2021-12-25 by Elkin-Vasily