PrintMakerLab/mks-wifi-plugin #250: File name is too long to upload. Please rename it.

Sabonius, 2021-12-04:
### Plugin Version

1.2.6

### Cura Version

4.12.1

### Platform

Windows 7 SP1

### Printer

Flying Bear ghost 5

### Reproduction steps

1. Connect to printer
2. MKS WIFI settings: "screenshot support" - "enabled"
3. Send to print via button "Print over flying bear ghost 5"

### Actual results

1. Get "File name is too long to upload. Please rename it." window.
2. Renaming doesn't help.

If "screenshot support" - "disabled" - works fine.

### Expected results

1. Printing commence!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Не знаю что добавить.

Elkin-Vasily, 2021-12-04:
Add a log-file please

Sabonius, 2021-12-04:
[cura.log.1.log](https://github.com/Jeredian/mks-wifi-plugin/files/7654483/cura.log.1.log)
Failed to reproduce the bug. Now everything works.
And files added do not contain times bug happened as far as I can see.

Do not know what to do next.

Elkin-Vasily, 2021-12-04:
Nice to hear that all is ok

closed 2021-12-04 by Elkin-Vasily