PrintMakerLab/mks-wifi-plugin #259: Printing not finish

feivur, 2021-12-22:
### Plugin Version

1.2.6

### Cura Version

4.11.0

### Platform

Windows 10

### Printer

FBG5

### Reproduction steps

After printing is complete, the plug-in remains in the "printing ..." state. It is not possible to stop printing, nothing happens after "Abort print" -> Yes. Have to restart the printer and cura.

### Actual results

After print finishes, plugin freezes

### Expected results

After print finishes, able to start new one

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Jeredian, 2021-12-23:
@feivur , please attach the cura.log file.

Elkin-Vasily, 2021-12-23:
What printer firmware do you use?

feivur, 2021-12-24:
[cura.log.txt](https://github.com/Jeredian/mks-wifi-plugin/files/7774014/cura.log.txt)
Чтобы плагин одумался, приходится переподключать его в настройках принтера в куре и много раз жать кнопку остановки печати. И не всегда помогает.

Elkin-Vasily, 2021-12-25:
According to your log file you have got custom printer firmware that works with wifi module not correctly. So after printing is finished wifi module tells that printer is still in "PRINTING" status. Just change the firmware, or make an issue in firmware's repository.

closed 2021-12-25 by Elkin-Vasily