PrintMakerLab/mks-wifi-plugin #260: 1.2.6 Remove function does not work

UncleJessy, 2021-12-31:
### Plugin Version

1.2.6

### Cura Version

4.12.1

### Platform

MAC oS

### Printer

Neptune 2

### Reproduction steps

1. Setup MKS Plugin
2. Assign IP address
3. Connect to Printer
4. Disconnect from Printer
5. Click Remove

### Actual results

When attempting to Remove a connection entry nothing happens. 
If you Restart Cura and try to remove - it will not remove
If you uninstall the plugin - restart cura, reinstall the plugin, restart cura - you still cannot remove the wifi entry

### Expected results

You should be able to select a wifi entry and click remove and it remove the IP Address entry. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/7796660/cura.log)

Jeredian, 2022-01-09:
@UncleJessy , can you try the same thing but with latest version from develop branch? (you will need to replace files at the Library/Application Support/cura/4.12/plugins/MKSWifiPlugin/MKSWifiPlugin folder)
Because I tried for 4.12 and 4.13 versions on Cura and can't reproduce your issue.

Jeredian, 2022-01-22:
@Elkin-Vasily , please try to reproduce it too, because I can not. If it does not appear for latest version of the plugin, let's close this issue.

Jeredian, 2022-02-06:
closed due inactivity

closed 2022-02-06 by Jeredian