PrintMakerLab/mks-wifi-plugin #269: Crash Cura when printer connection state changed

nallath, 2022-01-31:
### Plugin Version

1.3.0

### Cura Version

4.13.x

### Platform

Any

### Printer

Doesn't matter? 

### Reproduction steps

No idea; We only get the crash reports. 

### Actual results

Cura crashes. 

### Expected results

No crash. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://sentry.io/share/issue/32dff18b926346d2a7b4d61729276927/ 

There is more information about the context of this crash but that isn't visible in the publicly shared link. Please contact me if you need more info. 

Over 60 different users seem to have had this crash.

Elkin-Vasily, 2022-01-31:
Attach a log-file please

nallath, 2022-02-01:
I pulled some of the breadcrumbs from one of the crashes.

```[MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [816]: _send_data: M105  M997  
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: ok
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: M997 IDLE
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: ok
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: T:26 /0 B:27 /0 T0:26 /0 T1:26 /0 @:0 B@:0
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: T:27 /0 B:25 /0 T0:27 /0 T1:27 /0 @:0 B@:0
[MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [816]: _send_data: M105  M997  
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: ok
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: M997 IDLE
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: ok
[MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1003]: mks recv: T:26 /0 B:25 /0 T0:26 /0 T1:26 /0 @:0 B@:0
AttributeError: 'NoneType' object has no attribute 'getKey'
```

The issue as far as I can see from looking through your code is that self._current_printer is of type Optional[MKSOutputDevice
]. If you want to catch issues like this in the future, I can recommend using mypy typing, since it would warn you that one of the types that the variable can be doesn't have the "getKey" function.

Jeredian, 2022-02-02:
@nallath, good catch! MyPy is a good idea, I never thought that it will be needed for our small plugin before.
I will try to provide a new version with this fix till the end of the week.

nallath, 2022-02-02:
You don't _have_ to use it of course, but I've found that it has benefits for small & large projects alike. It takes a bit of time to get used to, but the extra effort pays itself back in my experience.

chris-red, 2022-02-05:
I have this issue along with another. The Printer will often connect then disconnect when starting Cura. I have found if I remove and Re add the ip it 'fixes' it but cura also crashes when I do this. it has only been since the last update.

info from the crash...

Traceback (most recent call last):
  File "C:\Users\Chris\AppData\Roaming\cura\4.13\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevicePlugin.py", line 157, in _onPrinterConnectionStateChanged
    if self._current_printer.getKey() != key:
AttributeError: 'NoneType' object has no attribute 'getKey'

Thread 0x00003aac (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000fb4 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000039bc (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001d50 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003468 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.13.1\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003c30 (most recent call first):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003898 (most recent call first):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001864 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000024cc (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000007b4 (most recent call first):
[... 43 lines cut ...]
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x00001fa0 (most recent call first):
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 398 in _logInfoWidget
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 184 in _createDialog
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 88 in __init__
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 162 in exceptHook
  File "C:\Users\Chris\AppData\Roaming\cura\4.13\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevice.py", line 603 in setConnectionState
  File "C:\Users\Chris\AppData\Roaming\cura\4.13\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevice.py", line 857 in printer_set_connect
  File "C:\Users\Chris\AppData\Roaming\cura\4.13\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevice.py", line 998 in on_read
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 461 in _show
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 482 in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1100 in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 461 in _show
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 482 in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1100 in event
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 396 in exec_
  File "C:\build\Cura\4.13-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 877 in run
  File "C:\build\Cura\4.13-exe\build\inst\bin\cura_app.py", line 236 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

awakenedlast, 2022-02-06:
I'm having the same issue as chris-red. The printer will repeatedly connect/disconnect. I will try his work-around until then.

chris-red, 2022-02-06:
Have reverted to 1.2.6 and it is fine, solid as it was before.

Jeredian, 2022-02-06:
@chris-red, yep this is the current workaround for the `connection state changed` issue.  We working on the solution, and the plugin will have an update asap.
About `repeatedly connect/disconnect`, you need to disable WiFI settings support, restart Cura, and enable it again. This workaround will clear olf IP values and prevent the race condition.

chris-red, 2022-02-06:
Thanks @Jeredian and thanks for maintaining this, its a great bit of software. I love the features I get from it and a cheap WiFi module as made my printer much better.

Jeredian, 2022-02-06:
`Repeatedly connect/disconnect` issue was moved to the separated task (#271)

PR #270 (2022-02-06): [MWP-269] Crash Cura when printer connection state changed
closes #269
closed 2022-02-06 by Jeredian