PrintMakerLab/mks-wifi-plugin #282: How to disable monitor tab "auto-click"?

molor, 2022-03-11:
### Is your feature request related to a problem?

I'm always frustrated when Cura switches to monitor tab that's useless _for me_ in 90 % of time, so it's just the need to make another click every time.. I think that there was no such behavior in early versions of plugin, and it was very convenient for me and my team — idk why it's changed and for what.

### Describe the solution you'd like

I hope that you can add a checkbox to plugin settings that can change this and keep the active tab without change.

### Describe alternatives you've considered

Continue to click on previous tab every time after data was sent..

### Affected users and/or printers

Plugin version 1.3.1
Cura version 4.13.1

### Additional information & file uploads

Thanks

Jeredian, 2022-03-22:
@molor , looks like a good idea, we will try to implement this feature asap.

ParkerK, 2022-05-22:
@Jeredian @Elkin-Vasily

I would be happy to try to implement this for you if you'd like. How would you feel about adding a third tab for additional settings?

[image]
[image]
 
I'm happy to discuss more if you'd like, I don't want to step on any toes or skip any steps you may have for community PRs :)

Jeredian, 2022-05-23:
I think file remaining and max length can be moved too, because they are not related to the network settings in that case.
@Elkin-Vasily , WDYF ?

Elkin-Vasily, 2022-05-23:
> I think file remaining and max length can be moved too, because they are not related to the network settings in that case.
> 
> @Elkin-Vasily , WDYF ?

Yeah, sounds right.

Jeredian, 2022-05-25:
@ParkerK , feel free to make PR, and if it's not difficult for you, move also `file renaming` and `max length` to the additional settings tab :-)
I created a new branch for this task ( `issue-282` ) you can use it as target instead of develop branch. 

@Elkin-Vasily, as I remember we will need to update translations too, because there will be a new label for checkbox and 2 other will be moved to new tab. I think we can make it from our side, after PR will be done.

Elkin-Vasily, 2022-05-25:
> @Elkin-Vasily, as I remember we will need to update translations too, because there will be a new label for checkbox and 2 other will be moved to new tab. I think we can make it from our side, after PR will be done.

Sure

PR #317 (2022-06-10): [MWP-282] Monitor Tab autoopen setting added
Advanced settings tab added

closes #282
closed 2022-06-10 by Jeredian