PrintMakerLab/mks-wifi-plugin #286: Button "Delete" is not work, files from SD card cannot be removed remotely

WEREWOLF1973, 2022-04-23:
### Plugin Version

1.3.3

### Cura Version

4.13.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Try all plugin releases 

### Actual results

All plugin releases have issue with inactive button "Delete file" in SD card menu

### Expected results

Possibility deleting files from SD card from plugin menu

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

ParkerK, 2022-05-18:
@WEREWOLF1973 can you add a log file?

WEREWOLF1973, 2022-05-18:
> It _might_ be due to the reconnect/connect and disconnect loop bug #289
> 
> Your logs show a constant attempt to reconnect, and if Cura is disconnected it may not allow you to delete (since it thinks the connection is broken).
> 
> You can either wait for an update, or you could try editing this file https://github.com/Jeredian/mks-wifi-plugin/pull/290/files
> 
> It will be in `C:\Users\YOURUSERNAME\AppData\Roaming\cura\4.13\Plugins\MKSWifiPlugin\MKSWifiPlugin\MKSOutputDevice.py`

Done, but no effect....

WEREWOLF1973, 2022-11-09:
No any progress till now

PR #356 (2022-11-15): [MWP-286] Delete/print file fix
closes #286
closed 2022-11-15 by Jeredian