PrintMakerLab/mks-wifi-plugin #287: Support new Cura 5.0 updates

Jeredian, 2022-04-25:
### Is your feature request related to a problem?

-

### Describe the solution you'd like

-

### Describe alternatives you've considered

-

### Affected users and/or printers

-

Jeredian, 2022-05-13:
Short update: Develop branch updated via PR from @ParkerK , with Qt6 migration changes.
On **Monday** I will test it on OSX and Windows, and if all comes ok will provide new release.
Since these changes our plugin will support only **Cura 5.0** and newer.

I will try to keep in Cura Marketplace _1.3.2_ version only for 4.*+ Cura, but I'm not sure that is possible because of inner publish process. Anyway, I will keep you informed.

kadorito, 2022-05-17:
Hello, I can't find any version on cura marketplace currently, is this normal ?

Jeredian, 2022-05-17:
@kadorito , yes, because we didn't released support for Cura 5, yet. 
But if you use Cura 4.* the 1.3.2 should be available for you, if it's not - please create a separated ticket.

kadorito, 2022-05-17:
I'm using cura 5, I understand why then :-)

ParkerK, 2022-05-17:
> Short update: Develop branch updated via PR from @ParkerK , with Qt6 migration changes. On **Monday** I will test it on OSX and Windows, and if all comes ok will provide new release. Since these [...]

PR https://github.com/Jeredian/mks-wifi-plugin/pull/294 will make it so 1.3.3+ will support both 4 and 5 if you want to go that route :). if not, feel free to ignore/reject

iqubik, 2022-05-18:
Are looking forward to. And the bug with cyclic disconnection / connection of the printer on the old version has not gone away.

closed 2022-05-25 by Jeredian

tpimh, 2022-05-27:
Just updated to Cura 5.0, and couldn't install the plugin.

https://marketplace.ultimaker.com/app/cura/plugins/Jeredian/MKSWifiPlugin => "The given author does not exist."

tpimh, 2022-05-27:
Thank you for clarification! As I mentioned, manual installation worked perfectly.

PR #293 (2022-05-25): [MWP-287] Support new Cura 5.0 updates
closes #287
closes #299