PrintMakerLab/mks-wifi-plugin #291: Not working on Cura 4.13.1

AgnidDrage, 2022-05-13:
### Plugin Version

1.3.2

### Cura Version

4.13.1

### Platform

Windows 10

### Printer

Hellbot Magna 2 230

### Reproduction steps

1 - Install correctly the plugin
2 - Open Cura
3 - Cura cannot load plugin 

### Actual results

Cura cannot load the plugin

### Expected results

Cura loads the plugin and I can use it

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8689339/cura.log)

Jeredian, 2022-05-13:
It's not a bug, because you installed the plugin in incorrect way.
The correct folder for plugins is cura plugins folder inside **AppData**, f.x. `\AppData\Roaming\cura\4.13\plugins\MKSWifiPlugin`

Also for I recommend to you to install plugin via Cura Market, instead of manual installation. And only if you need not released version go to the  cura-plugins-folder and replace files them ( without any kind of extra folder creation)

[image]

Jeredian, 2022-05-13:
Since next release we will try to support *.curapackage , to prevent user to manually install custom version. It will keep us save out of this type of issues

closed 2022-05-13 by Jeredian