PrintMakerLab/mks-wifi-plugin #314: File name lengt

snow79, 2022-06-01:
### What is your question?

Hi all and so sorry for my question...
I have Fling Bear Ghost 5 like 3d printer, with board Mks Robin Nano V3, Marlin last update. now i use Cura 5 and i have this problem:
Problem 1:
When i send file from cura 5 to my printer using Wifi the file name change and go as FBG~1.
don't accept long file name;
Problem 2:
some times lost connection and re-connect endlessly, for solve this problem i have add a new printer and if i switch to other printer and return to ghost 5 the connection is ok
Problem 3:
the previews on cura 5 don't function 

The problem number 1 and 2 are same with cura 4 and cura 5; i have try with 2 mks board with marlin firmware (board version 1.2 and board version 3.1)
the problem number 3 it's only with cura 5 and not with cura 4

With the original Fling Bear Ghost 5 firmware i don't have any of this problem, but i need this firmware because:
i use driver 2209; i have enabled lineadvance, i have 3dtouch

The only my real problem is the file name on the printer... 

what can it be caused by?

Elkin-Vasily, 2022-06-06:
1) Your printer firmware shortens file names when they uploaded via WiFi
2) Attach a log-file please
3) Try latest plugin version from the market

Elkin-Vasily, 2022-06-20:
Closed due to inactivity

closed 2022-06-20 by Elkin-Vasily