PrintMakerLab/mks-wifi-plugin #315: Cura 5.0, MKS Plugin 1.4.0. Cannot connect to machine.

FedorSosnin, 2022-06-04:
### Plugin Version

1.4.0

### Cura Version

5.0

### Platform

Mac OSX 13.2.1 (M1 Max)

### Printer

Kywoo3D Tycoon

### Reproduction steps

1.Install app via .curapackage file
2. Open plugin (After restarting Cura)
3. Add IP address
4. Click "Connect"

### Actual results

Nothing happens after clicking connect. Tried reinstalling both manually and via package. 

### Expected results

Connect to plugin

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

Elkin-Vasily, 2022-06-06:
Ok, attach a log-file please

iqubik, 2022-06-08:
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8862287/cura.log)

FedorSosnin, 2022-06-08:
> Ok, attach a log-file please

Sure thing. Here is mine:

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8862618/cura.log)

iqubik, 2022-06-08:
Так же чистый лог. Без дополнительных плагинов. Несколько запуско
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8862785/cura.log)
в

iqubik, 2022-06-11:
У вас написано что плагин для mks tft 4.0 а на flsun q5 экран меньше. И в последней версии и в прошлых теперь вообще ничего не работает. Ни в 4 куре ни в 5 ни на одной из версий ничего не работает.

iqubik, 2022-06-13:
Здесь кто нибудь есть:?

iqubik, 2022-06-15:
Я разобрался. Вроде работает. Сбросил виндовс к заводу. Думаю дело было в настройках приоритета траффика в windows. Поднимал ubuntu на windows 10 через wsl2 и там был ряд команд через powershell по работе с пакетами. Думаю что эти команды как то могли нарушить работу плагина.

iqubik, 2022-06-16:
Нет, после первой же перезагрузки опять вообще никакой реакции. Даже нет сообщений о реконнекте. Просто тишина.

iqubik, 2022-06-16:
нашёл интересную закономерность. Создал нового пользователя в windows 10 и поставил 5 версию с плагином из магазина - работает, а под основным пользователем нет.
Прилагаю оба лога с идентичными конфигурациями - чистая установка + плагин. Первый запуск

Не работает ничего:
[NWcura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8919692/NWcura.log)

Всё работает:
[Wcura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8919693/Wcura.log)
.

iqubik, 2022-06-16:
Проблема была в Lantern VPN. Даже когда просто иконка отключенного Lantern висит в трее плагин не будет работать. Lantern - выход - всё ок.

FedorSosnin, 2022-06-16:
Interesting find...for Windows.

The problem still exists with or without VPN on OSX 13.2.1.

Elkin-Vasily, 2022-06-19:
> Interesting find...for Windows.
> 
> The problem still exists with or without VPN on OSX 13.2.1.

Try to use this version: https://github.com/Jeredian/mks-wifi-plugin/tree/issue-316

I cannot reproduce this bug, so I need some help with testing.

PR #321 (2022-06-21): [MWP-316] Continuous reconnection fix
closes #316
closes #315