PrintMakerLab/mks-wifi-plugin #316: Continuous connection and disconnection to WiFi

Fulvio1, 2022-06-06:
### Plugin Version

1.4

### Cura Version

5.0

### Platform

Windows 10

### Printer

Flsun Q5

### Reproduction steps

Installing new version of MKS WiFi for Cura 5.00 from the market.
Reboot Cura

### Actual results

The connection is not stable. It connects and disconnects all the time making it impossible to use!

### Expected results

A stable connection with the printer!

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8844716/cura.log)
[image]

snow79, 2022-06-06:
Same problem:
Plugin Version
1.4

Cura Version
5.0

Platform
Windows 10

Printer
Flying bear Ghost 5  \\ Board Mks mini nano v3.1

Reproduction steps
Installing new version of MKS WiFi for Cura 5.00 from the market.
Reboot Cura

Actual results
The connection is not stable. It connects and disconnects all the time making it impossible to use!

Expected results
A stable connection with the printer!

- [x] Cura log file 
- [x] Screenshot  
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/8845840/cura.log)
[image]

laurent7878, 2022-06-11:
same issue too. When i uninstall plugin and add beta version it works again.

Elkin-Vasily, 2022-06-19:
@FlyDrag thanks for your research it helped a lot.

Try to use version from this branch: https://github.com/Jeredian/mks-wifi-plugin/tree/issue-316

I cannot reproduce this bug on my setup, so i need some help with testing.

FlyDrag, 2022-06-20:
@Elkin-Vasily I checked this branch. It seems that this issue was fixed.
I'll keep running issue-316 branch for some time.

PR #321 (2022-06-21): [MWP-316] Continuous reconnection fix
closes #316
closes #315
closed 2022-06-21 by Jeredian