PrintMakerLab/mks-wifi-plugin #320: The setting button is gone.

runfortop, 2022-06-19:
### What is your question?

Hello. I've been using your great program, but I've run into difficulties that I can't use.
After adding a new 3D printer in Cura program, I tried to connect to WiFi, but the button for WiFi setting was not created. Then I went back to the original 3D printer and tried to check the WiFi connection, but the button for setting disappeared. So, even though I deleted and installed both the plugin and the Cura program again, the button for WiFi setting was not created. How can I create a button to use this plugin?

Elkin-Vasily, 2022-06-19:
Attach a cura.log please

runfortop, 2022-06-19:
I am attaching the stderr.log here.

[stderr.log](https://github.com/Jeredian/mks-wifi-plugin/files/8936588/stderr.log)

Elkin-Vasily, 2022-06-20:
From your log: Plugin [MKSWifiPlugin] has been disabled. Skip loading it.

Enable the plugin.

For this:
1) Open Marketplace window;
2) Click gear icon;
3) Click enable button on MKS Wifi Plugin.

runfortop, 2022-06-22:
Success.
Thank you very much for your help. But I figured out why this was happening. The enable button you inform me is displayed with the opposite meaning in Korean. So I couldn't solve the problem.

thank you