PrintMakerLab/mks-wifi-plugin #340: Printer reboots when printing starts

PingWIn87, 2022-07-15:
### Plugin Version

1.4.0

### Cura Version

5.0

### Platform

Windows 10

### Printer

Flying Bear Ghost 5

### Reproduction steps

1. slice model
2. click "Print on Flying Bear Ghost 5" button 

### Actual results

the printer is rebooting

### Expected results

start printing

### Checklist of files to include

- [X] Log file

### Additional information & file uploads

[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/9118470/cura.log)

Elkin-Vasily, 2022-07-15:
Hello. Is this a file you try to print: FBG5_Bed_Level_test_Speed_Print_Test.gcode ?

If so - try to use shorter names, it could be that your printer firmware tries to shorten this name and something goes wrong.

PingWIn87, 2022-07-15:
> Hello. Is this a file you try to print: FBG5_Bed_Level_test_Speed_Print_Test.gcode ?
Yes.

> If so - try to use shorter names, it could be that your printer firmware tries to shorten this name and something goes wrong.

ОК. I'll try.

Elkin-Vasily, 2022-07-26:
Is this issue still actual for you @PingWIn87 ?

Elkin-Vasily, 2022-07-29:
Closed due to inactivity

closed 2022-07-29 by Elkin-Vasily