PrintMakerLab/mks-wifi-plugin #343: Hellbot Magna SE 32 Bits Error

kaakeee, 2022-08-18:
### Plugin Version

1.4.0

### Cura Version

5.1

### Platform

Windows 11

### Printer

Hellbot MAgna SE 32 Bits 

### Reproduction steps

1. Open a .stl in cura 
3. Configure MKS in Default 
3. Create the file in Default (the only one that works) 
4. When entering it in the printer, the image runs

### Actual results

smudged or damaged image

### Expected results

viewable image ok

### Checklist of files to include

- [X] Log file

### Additional information & file uploads

![image](https://user-images.githubusercontent.com/111094495/185379573-11563405-105b-401f-a7cd-920b50925c15.png
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/9372994/cura.log)
)

Jeredian, 2022-08-23:
@kaakeee , did you tried to setup preview image settings?
You can find this option at the plugin settings window.

kaakeee, 2022-08-23:
Hi, yes, i proved all options in the complement, default, FLSUN QQ-S, Flying Bear Ghost 4s/S,etc... and very options in custom, 100 200 300 50...

Jeredian, 2022-08-23:
@kaakeee , I guess you can try in `custom` mode with any other values, like 25, 30, etc. Don't forget that yuo may need to have different values for `simage` and `gimage`. If you find values that will fit for your printer model, please back to us, and we can add into list. 
We can't setup it by ourself because we doesn't have this printer model.

kaakeee, 2022-09-12:
I reinstall complete with GitHub code, not install with Cura Market, sorry for misunderstanding.

but it disables the layer fan for some reason, but well...

closed 2022-09-12 by kaakeee