PrintMakerLab/mks-wifi-plugin #345: Window for response to a command

bellzw, 2022-08-29:
### Is your feature request related to a problem?

Not really a problem or error.  It would be nice if the user could see the response to g-codes sent to a connected printer.  For example, the response to M503 contains the current working speeds, accelerations, jerks, etc.

### Describe the solution you'd like

A window, with a vertical scroll bar, in the ample empty space under the Printing Time info.
[image]

### Describe alternatives you've considered

Looking in Cura's stderr.log, but I do not see the response to the M503.  Here are the lines I do see (the relevant lines are bolded):
2022-08-29 17:33:16,175 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: T:23 /0 B:24 /0 T0:23 /0 T1:24 /0 @:0 B@:0
2022-08-29 17:33:17,995 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [825]: _send_data: M105  M997  
2022-08-29 17:33:18,010 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: ok
2022-08-29 17:33:18,064 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: M997 IDLE
2022-08-29 17:33:18,064 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: ok
2022-08-29 17:33:18,079 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: T:23 /0 B:24 /0 T0:23 /0 T1:23 /0 @:0 B@:0
**2022-08-29 17:33:19,989 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [825]: _send_data: M503  
2022-08-29 17:33:20,004 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: ok**
2022-08-29 17:33:22,013 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [825]: _send_data: M105  M997  
2022-08-29 17:33:22,035 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: ok
2022-08-29 17:33:22,082 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: M997 IDLE
2022-08-29 17:33:22,097 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: ok
2022-08-29 17:33:22,097 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.on_read [1013]: mks recv: T:23 /0 B:24 /0 T0:23 /0 T1:24 /0 @:0 B@:0

### Affected users and/or printers

Everyone, I'd expect.

### Additional information & file uploads

This is the entire stderr.log file in case I missed the response to M503.  stdout.log is empty.

[stderr.zip](https://github.com/Jeredian/mks-wifi-plugin/files/9448127/stderr.zip)

I do know that the printer responds to M503 because I get this from Pronterface when I connect via USB:
>>> m503
SENDING:M503
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z400.00 E93.60
echo:Maximum feedrates (mm/s):
echo:  M203 X350.00 Y350.00 Z2.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1100 Y1100 Z100 E5000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1100.00 T1100.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.30 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P23.89 I1.37 D104.50

Jeredian, 2022-08-31:
@bellzw , thanks for feature suggestion. It looks really interesting idea, I guess we can try to implement it.

@Elkin-Vasily ,do we have any blocker moments for this ?

Elkin-Vasily, 2022-09-19:
MKS Wifi Module does not transport every command that it gets from printer main board. So wifi connection and USB connection are not the same.

We are using Cura's form for our Monitor tab, and there is no such dialog.

There is no place for this frame with small resolution monitor. For example:
[image]

closed 2022-09-19 by Elkin-Vasily