PrintMakerLab/mks-wifi-plugin #350: Failed to start print after upload

enishi1357, 2022-09-17:
### Plugin Version

1.3.2

### Cura Version

4.11

### Platform

Windows 10

### Printer

Dovov S2

### Reproduction steps

1. Slice file
2. Click Print over Dovov S2
3. Finished uploading
4. Reached nozzle and bed temp
5. Printer does not move
6. Waited for a long while until temp drop

### Actual results

Printer did not move for example suppose to home axis and auto level with 3d touch

### Expected results

Printer should print since it uploaded the file into sd card and the printer can print off sd card no problem.

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

none

Elkin-Vasily, 2022-09-17:
Try to turn preview off. Does print starts correctly then?

enishi1357, 2022-09-17:
Do you mean preview in cura after slicing? How do you turn off preview?

Elkin-Vasily, 2022-10-21:
> Do you mean preview in cura after slicing? How do you turn off preview?

Manage printers -> MKS WiFi Plugin -> Preview settings -> Screenshot support (checkbox)

closed 2022-11-26 by Elkin-Vasily