PrintMakerLab/mks-wifi-plugin #351: Disconnect and reconnect multiple times

enishi1357, 2022-09-17:
### Plugin Version

1.4.1

### Cura Version

5.1.1

### Platform

Windows 10

### Printer

Dovov S2

### Reproduction steps

1. Slice file
2. The box shows save to sd card then flash to print over Dovov S2
3. Click print over dovov s2

### Actual results

Shows progress uploading file but failed halfway

### Expected results

Complete uploading

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

Updated wifi module firmware

Elkin-Vasily, 2022-09-17:
Add a log file, please

enishi1357, 2022-09-17:
what is the directory to log file?

enishi1357, 2022-09-17:
[cura.log](https://github.com/Jeredian/mks-wifi-plugin/files/9592685/cura.log)

Elkin-Vasily, 2022-10-21:
Do you have only one instance of Cura opened at the same time?

Printer socket can work only with one connection at the same time.

ivanov-alex, 2022-11-04:
I had the same issue.
Faced it when updated to Cura 5.2.1 from 4.*
Printer constantly connects and disconnects. In a single instance of Cura.
Resolved for myself by downgrading to 5.1.0, it works for me without any glitches. Was not trying other versions in between.

Nixellion, 2024-01-03:
I've had this issue again, and indeed, restarting my PC and printer worked as a fix. It seems like if you open and close Cura, sometimes the python process might be still working somewhere in the background, trying to reconnect to the printer.

Maybe a way to detect and kill old processes when a new is starting or some kind of lock mechanism should be able to help fix or mitigate this issue.

Jeredian, 2024-02-08:
@Nixellion, thank you. That sounds interesting, we need to delve deeper into this idea and investigate the possibility of closing the socket along with closing Cura. 
During a debug it usually completes, but maybe something different happens when you change versions.