PrintMakerLab/mks-wifi-plugin #352: No option to save as TFT

mattinflorence, 2022-11-03:
### What is your question?

With Cura 5.2.1 and the latest version of the MKS plugin, there is no longer the option after a file is sliced to "save as tft"?

Elkin-Vasily, 2022-11-03:
In the newer plugin versions (if the preview option activated for the current printer) preview will be added to the default "save as file" option.

mattinflorence, 2022-11-03:
> In the newer plugin versions (if the preview option activated for the current printer) preview will be added to the default "save as file" option.

I've confirmed that an eariler version of the plugin, creates a different output of the image data and a different format...

for instance, this is the format that shows the preview image... (its long so my apologies) and this is cura 4.8 with plugin 1.0.2

;gimage:0`000<P000380000?<?M1CH1001_9P0000000000000006KE9~f6gFF~A]F5]:Jd1~GVa:On97_6_4F~ioj7oVF3IohWmRFT2?oUVjF;A~g5TlOna^FV_4FTJOl6cJARYKC]o`Ofi^g^o`FLAG]8ojKM14Y4D^Jla6ZViNAZjom7o^coi7;UTbSoI]g:oj[o1onS>NooYI=S<FAJ1nj9o`<Q`aP5Y4WohaPS:@K^A7]5Po3oi^D4LhF~QH_4@HF3Pc64HT<YPcV:ohAJ`d47iTB395;;od<aI~fU_<Flhd66eDOfA5X]mjW^i4TVm~KMImE7kXS^Y6YWmRcg0d[Yk^=99nkT@JS^c?o8mPWg9nH6mVFd94Z242=BI8<VaN<PhQPR8GDZHS50:@KVJ>KWgJKEi7ZRJZOEK?LYmn91@e8R@RAcQoHWgRK^@d[3DHBDYJ`0n0240?l0o`3o0?l0o`3H3A80Z@d~i3[P09D=>@29?0d_A2oQ<>0=43`0P0d_DJlbB0dB07P]12cQ:><Qh0l;DVPY11@0LN@ih0@Zhc7P3a1bZ0@F06~49n8Zhb7P3aGR9^1bb>4[h0@G06H49>4Xh<X?3eZ1hHdch0P711P0H@@Phb3P2Pd?2N4Sh04;hHgPB`P811T0G@@K=n4Xh0XCCn4nhR[P0A3QSN1[hROPb3hV11T0F@@FhekP2QEB0AOQSN2[;HQn11T]05D44bT:5^<Qh04MhHgP2`M8_RH46@1Bi3SP10gR?^0:5S4S0B8UhHgP2`Pc_T846P1?i3OP10Q62QH_0BM5hHgP2`WR9n2NPP@I04d42><~h0XC@`4ZQN6=h0~9hROQA^0n`^8mh0@H04_Q:^042BTn2P`S0BnUhHgP2`[R9n56h087:0@H04WQ:^042bI9hbOPB^4]=N01=<GQSN0;2^8WhHKP0PL~11H]04OQ:^043RT]hGGP0C@51n6=h0~:hROQY^0220@FhS7P04GQ:RcP10oS?N5Eh04d1@S
;gimage:QSN0;2n8WhLKP0PP45N8ah013hBX~h2gR313P:04d1@WQSN0;2n8Wh@H7hSkP0PMVhP`A:c7T=^00@>4m;>0[1Q<1<@D;hHgP2`_R9n462N0220HBhT~a=^00?^4Z;>1;1Q@1;`D<hHgP2`cQ1P[P9P871Q4]hT~a=N00?>4m;>1;hROP1QDR0B`53N58CN0;3>8Wh@H;h0881PlT<>9;<N<lh00ii3GR=N4~h6~f1QL1:PD>hHSP2`cR9n463>0`0PLX1PgR;Dm;<CWP03SQC>1[ERd65`4W1A3QR>0;3>8Wh@H>:n021`H<=N8]Cd]Ah00fhSGQ;>1[M^4Uh0HH0BD54N68h0~=hROQ1PoPXULR1PTT=N8]Kd~a=N00=><[hBcPBn8Wh7KR8^0c1QX18PDBhHSP2`gR9n464>0VHYL627GQ<n9_Bc7P03<ohBcPJgKQ9N1c1QX180DChHSP2`gR9n464>98h29ghD3P1PREhRe_Bc7P03;Q;>1;hSSPMSFC1Q~17PDDhHSP2`kR9n463n0c90H=A9DhhVm;hBcP033Q;>1[M^4kh9<67BT16PDFhHSP2`kR9n462RGP1Q:4UN8]Kd_P031;hSSPMSFC1R0160DGhHSP2`kRAn5Vh3D65^4P:n2TUCSRKd_P02m;MSRCY^4Ph2P66`4F1ASQR>0;3SP66^4PJn2TUN8]Cn0`hR_P02h[hROPM^4kh9>VhB3PJ3~66RT14PDJhHSP2`Pf1QoQbn2TUN8]Cn0`02h[UY<lY~S6h@0Dh2T140DKhHSPJn8]h@0Lh<HRAN6[h2bTUN9_h00^hRSPMY<lY~QQh@0Jh04>1AcQ=N0Qh@0VhSkPYN6[h2bTUN8];n00;^8Ph5JcY~S1h@0Ih2D12`DH=b;Q02WR?N4Xh<GQbn34UN8_h00^hR3P=[>V:JPR`E[QP>075@491A@Y1bWQP2Q]:>35hJ_P;:@cU@0^hR3P~lK80@NJ8@LG`DGSQN8ah0D:=c~7;n
;gimage:4X[BSPaCOQZn0~Y3=EhCWP02kR8>0fLlK80@LJ1c8j1aEQhBWRAG?Q>N:I<N4Ph2`k1c7Q;GkR9^7=:>358^6[h2bT<cGQ?>00;^8PhCWPDlK88~4J1n8Zh9;Q;N074bcRIN0fhRgP:n8ahBkP?eD7<bWQg^8VhB>]B>2U=bSQRn1<Y>8]hCcP02kR8>4i:^36b2811aX74PPa>R4743`V90LfhBf`g^8VhLdXh2>U=n6[h2cR?N24hbGQ?>00;n5Ih:H~b0476POR9^0B1m7Q:>0~1d7Q:>0_hA08g^96hJe8h<DghH_PC2I4hBWS9N4lh00_hETZ9N16;<P11aX84PPA28[T=><Ph0Ld=n<Ph0l8hA08g^8VhB>]B>2UEn6;h4`VhDWS9N4lh00_hEU:h4K89@476POR9^0B1n4`h1482PQ^@A~W8AT7:Pl7hA08g^96hJe8h2>U=bWQJn1<<>59G>00;n5IJR3Pb0476PPB21482PT>21@9i3?R70cQIe@H2>0I2BX?1c7Q40RnhTKQ8je8h:EG:N5[h5oQBEcP02oQFFX^hSgPR0476PPB2>4`h1482PT>21@9h@LA=1P8h2hI24X?1n4@2=kRA^6]J>2UEn5;h7oQBEcP02oQFFY^h4P121X84PPA20X93PTD1n5D1a8H2N0I24X?1n4@2MkRA^6]J>25EbROhDULh00_hEUZS^0X0@LJ21884@T:2@h9e>5d1a8H2N0I2DX?1n4@2=kRA^6]J>25=n56h7oQ?DULh00_hEUZSS;P<:4J2>8Vh1884@T:2@h9U>6d1a8H2N0I2DX?1n4@2KkRI^6]J>15<>5Vh7oQ?DTlhR7P02oQFFZ><^1@=V4J2188hC3P4@P:2@h:E>4D1`LB60WP6@UZcc7Q40SNhVKQSHRVh7oQ?DTlhR7P02oR<n4iJZkPT>4kh1X8hRKP4PPA2@X93PUDhA@81a8H2N0^6@U:3`OQ40VnhVKQ[N0lF>6Vh7oQ?DTlh
;gimage:00`hS?Q>FZ^h;0e<mXB2A48;`X93PWQ50T751P9h1T9J~lahA08g^9VhDgP99SQY^1ohCe9?>00<>8chCUZ[^2`UN4Uh5XB2A492PTi3PWQ50TWhT?P7@OQ1`XH2N0I2TX?1n4@2KkRA^0RhC?P^3KQY^1ohCe9?>00<>8chCUZ[^2`UN5Uh1894@T:2@h:hA@9h1d:h@L:h3CQ60SP6@YZcn4@2KkP?UcQ<n2h=^6Vh7oQJCcP033R<n4iJ^<[hHkP~;GQAN1T8i8A2Bl:2@h9hCPD2N0M2^472^1dhAP7h1T9JPl7hA09;N3LhC?P^3KQQRcPOn5Y?>00<>8chCUZhSGQS^2`eN4Uh<CQ;N0b4@T:2CT>2N4D2^0M2^472^2DhMSP6@YZcn7@8n0k9a`7f3KQQRcPOn5Y?>00<>8PhCUZhSGQS^2`5@L41bI<hRKP~Bl:2@h:hA@:h1d:h@L:h=CQV>0I2VX?1n4`hRKPVdLL1mPfhHH~h7oQJCcP033R8>4i:D[R=N6>h;0E1`@79~cQ<>0a2PX>2^4D2^0M2^472^0D1n5hh1T:RVlaB;]W70OH=^66;>1ohFTlh00`hR3Q>BU:hSGQS^2`5@L41bH<28;:3P[Q50[P7@[Q1`[P50WQ>>0I2RX_8PP7VhLL1mPfhFKR9n4~h7oQJCcP033R8>4i:D[R=N6>h;0E1`@79P`80PPghc7P3P[Q50[P7@[R@n472>0D2^4hh;WS<N0QI@P9NjLL1mSQQ^8WhBcPOn5Y?>00<>8PhCTYB^8ehHkP~1D710LV30P22BD11n58h3CQ50[P7@gRHn6Wh;CT<^=7h4<52PP8:E_770OHhFKRAn2OhFTlh00`hR3Q>BUZS^2`=]D420`80PTU0@T32S[R5`lZhJSP0`X52@P9:C~71a`7f>5VhROP:ioQBN8ohCcP033R8>4i:FZ>h;0fe@@830P22P490`_Q21GP0`X52@P9:@L870OHhFKR9n0[Wn59hSoQ
;gimage:?>00<>8PhEUZS^2`=]D420`80PX12@<;h@PEh0<:1@T82BT721`7f>56hROPBioQBN8ohCcP033R8>5IJXkP~3KE10P<208:0@T32n484^033@D920TY1`PL1mSQA^8Wh4^OhDWR?n4lh00`hR3QFFZ>h;0fe@@830P22P49hCGP0`[Q20cP0a<52@P9:@L870OHhBKR9n1[Wn59hSoQ?>00<>8PhEUZS^2`=]D420`80PX12N4eh0<:h@P:h0<E1@T82BT72=`cf>4VhROPJioQBN8ohCcP033R8>5IJXkP~3KE10P<208:0@WQ=N032^482^035@D920TY1`RlDmSQ9^8Wh6^OhDULh00`hR3QFFZ>h=3E10P<208:0@WQ=N032^482^035>4eh0D920TY1`RLLmSR9n2;Wn59G>00<>8PhEUZS^3@e@@830P22P4:0`[Q20[P0a;QMN052@P9:@L8W7?HhROPRioQBEcP033S>n5IJXkPd=D420`80PX12P<AhFSP0`oQeN052@P9:@L8O9?HZioQBEcP037QFFZ>h=3E10P<208:0@X38N4E1n0Q1@Pg20T72GbCf:^OhDTlhS;P037QFFZ>h=3E10P<23h22@4:0akQ5@OPP@D7E`P91`UL~mR[Wn59?>8bh00ahEUZS^3@e@@830QN0PQ50@P37^6Eh047YIL82@L893cCf:^OhDULh00ahS3Q>FZ>h=3E10P<27h21hG10akQ=N012VGG20T722@C1mR[Wn59G>00<N8`hCUZS^8hh;3E10P<22VNX~F10ah12bDG20P91`PT4`OHZioQBEcP037R<>4iJXkR>>2`eC`41``8:Kj21@TQ0ah12aL920PY1`OQ8>0T4`NhhC_PZioQBEcP037R<>4iJXkR>>2`eC`41``8:Ah7@PD:0ah12aL920M9an50h2@C1kSQ>n2[Wn59?>8Qh00ahS3Q>FX~K^8h [... cut ...] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
M10086 ;

thusly, when loading the print file from the 1.4.1 plugin created version, no preview image appears.  i can remove the M10086 from thr MKSPreview.py, then when loading the print file, the preview image is just a black image/square.

the string output from each version is vastly different

Elkin-Vasily, 2022-11-04:
Try to use the latest version from the devel branch of this repo

mattinflorence, 2022-11-04:
same thing...

Jeredian, 2024-02-08:
@mattinflorence , it looks like we've solved that in https://github.com/PrintMakerLab/mks-wifi-plugin/issues/366 . Please try checking with the new version. https://github.com/PrintMakerLab/mks-wifi-plugin/releases/tag/1.4.4 ?