PrintMakerLab/mks-wifi-plugin #365: No preview on Elegoo Neptune 3 Pro

jaegerschnitzel, 2023-01-06:
### Plugin Version

1.4.1

### Cura Version

5.2.1

### Platform

Windows 10

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

On my Elegoo Neptune 3 Pro no preview is shown. I tried with the following settings:

[image]

With Elegoo Cura it's and inbuild plugin version 1.0.2 it's working fine.

### Actual results

no preview on printer

### Expected results

preview shown on printer

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

[image]

boehser-enkel, 2023-01-07:
Same here. I used default values

ErockSaint, 2023-01-31:
I'm stuck in the same place on a Neptune 3 Plus. Doesn't seem to be much of a concern here though. Plugin version 1.4.2, set as active for the printer in settings, same preview settings as you have used above, and I don't get the save as tft option. Read somewhere here it was on the save file options(but that was a few versions back), I have about 5 options, none of them are tft. Save as tft works as it should, and shows a preview, when saving in Elegoo Cura.

xieyun32, 2023-02-08:
How use this plugin in Elegoo Neptune 3 Pro

finict, 2023-05-28:
@jaegerschnitzel @boehser-enkel @ErockSaint @xieyun32 @Torrosua 
Due to ELEGOO's proprietary encoding, a specific binary file is required to use the preview
Check my fork which works with latest Cura
https://github.com/finict/mks-wifi-plugin-N3

jaegerschnitzel, 2023-05-29:
@finict 
Thank you very much it's working fine :)

closed 2023-05-29 by jaegerschnitzel