PrintMakerLab/mks-wifi-plugin #366: No Preview on Elego Neptunne 3 Plus

Kungvald, 2023-02-07:
### Plugin Version

1.4.2

### Cura Version

5.2.2

### Platform

windows 11

### Printer

Elegoo Neptune 3 Plus

### Reproduction steps

I installed the plugin from the marketplace.

### Actual results

i did not get any errors but the abillity to save .tft file and ghet a thumbnail in the printer is not there.

### Expected results

I cant view picture of the print in my printer like if i use the elegoo cura slicer, where your older version of plugin is installed.

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

anyone working on a new version?

xieyun32, 2023-02-08:
How to use this plugin in Elegoo Neptunne 3 Plus

halbik, 2023-03-24:
What motherboard and screen does your printer have?

finict, 2023-05-28:
Due to ELEGOO's proprietary encoding, a specific binary file is required to use the preview
Check my fork which works with latest Cura
https://github.com/finict/mks-wifi-plugin-N3

Jeredian, 2024-01-23:
It looks like Elegoo uses the same encoding as Artillery (Sidewinder X3, etc). After a quick search, I found that this encoding can be achieved by a public .py version of the same lib. 
We will add support for both printer manufacturers in next plugin release.

Also for Elegoo users, I can suggest to take a look on https://github.com/Molodos/ElegooNeptuneThumbnails plugin.

PR #391 (2024-01-27): [MWP-366] feat: support model preview for Elegoo Neptune and Artillery Sidewinder series
Updated the list of printers in the preview settings 
- Artillery Sidewinder X3 (must support any other version of Sidewinder)
- Elegoo Neptune3+ (any versions of Neptune3)
- Elegoo Neptune2 | NeptureX (previous versions of Neptune)

Added support for custom encoders
- ColPic encoder, which was used in Elegoo and Artillery (with corrected variable naming)

[image]
[image]
[image]
[image]

closes #366
closed 2024-01-27 by Jeredian