PrintMakerLab/mks-wifi-plugin #371: Cannot delete/print file on sdcard  (drive_prefix for FlyingBear is null)

barban-dev, 2023-05-19:
### Plugin Version

1.4.2

### Cura Version

5.3.1

### Platform

Windows 10

### Printer

Flying Bear Ghost 4s

### Reproduction steps

I cannot delete/print a .gcode file stored on SD in the Monitor section 

### Actual results

File is not deleted and print job is not started

### Expected results

File should be deleted and print job should be started

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

I cannot delete/print a .gcode file stored on SD since M30/M23 commands are wrong for Flying Bear Ghost 4s/5 printers
Eg: M30 /filename.gcode should be M30 **1:**/filename.gcode 

According to the closed issue #358 drive_prefix is introduced to manage these issue but the string is null istead of "1:" even if I set the plugin's Printer Model parameter to "Flying Bear Ghost 4s/5".

I resolved by modyfing /qml/MonitorItem.qml in this way:
`Cura.MachineManager.printerOutputDevices[0].deleteSDFiles(drivePrefixTextInput.text, "", listview.model[listview.currentIndex])`
-->
`Cura.MachineManager.printerOutputDevices[0].deleteSDFiles(drivePrefixTextInput.text, "1:", listview.model[listview.currentIndex])`
  
and

`Cura.MachineManager.printerOutputDevices[0].printSDFiles(drivePrefixTextInput.text, "", listview.model[listview.currentIndex])`
-->
`Cura.MachineManager.printerOutputDevices[0].printSDFiles(drivePrefixTextInput.text, "1:", listview.model[listview.currentIndex])`

bobafetthotmail, 2023-11-13:
There is a "drive prefix" menu in the Open Files popup where you can select the 1:/

It's not explained that if delete fails you have to try different "drive prefix" though, I had to come here to this bug report to understand I would need to put 1:/ for the prefix in that box. 
I think adding a line like "if delete or print do not work try a different drive prefix here" would help a lot.

It works for me on a FlyingBear Ghost 5 
[image]

these are the available choices
[image]

bobafetthotmail, 2023-11-13:
opened a PR to add the description  #377

closed 2023-11-14 by Jeredian