PrintMakerLab/mks-wifi-plugin #376: Unable to Connect - Connect Button Not Wroking

mabbott093, 2023-11-03:
### Plugin Version

1.4.2

### Cura Version

5.3.1

### Platform

Windows 10

### Printer

Elegoo Neptune 2S

### Reproduction steps

I installed the wifi chip into the motherboard and followed the instructions until I had the assigned IP as shown on the printer, went to enter the IP address in Cura.

### Actual results

When I click the Connect button on the popup window in Cura nothing happens. It does not connect, no error messages. 

### Expected results

I was expecting the Printer to connect to Cura.

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

Everything is up to date, Cura, MKS plugin, and Printer firmware.

Jeredian, 2023-11-04:
@mabbott093 , could you please attach the cure.log file.
```
A log file for crashes and similar issues.
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
MacOS: $USER/Library/Application Support/cura/<Cura version>/cura.log
Ubuntu/Linus: $USER/.local/share/cura/<Cura version>/cura.log

If the Cura user interface still starts, you can also reach this directory from the application menu in Help -> Show settings folder
```

Is your 3D printer accessible within your wifi network? Use `ping <printer_ip_address>` to check it out.

Could you please share with us the name of wifi chip that was installed into the motherboard ?
What printer firmware name and version are you using?

nicf82, 2026-10-06:
I am having the same issue, my setup was working, then one day It just stopped seeing the printer, and as the OP said, clicking connect does nothing. I do have the symptoms:

`telnet 192.168.0.131 8080` connects successfully but then sending a command such as `M105\n` or `M105[Ctrl-M]` yields no response. Is this a fair test?

I am able to load the Web UI and see the interface for uploading .bin files etc, at http://192.168.0.131

The cura logs just repeat this:

```2026-10-06 13:48:26,813 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice._update [805]: MKS wifi reconnecting
2026-10-06 13:48:26,814 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.disconnect [233]: MKS socket disconnecting from 192.168.0.131:8080
2026-10-06 13:48:26,814 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.connect [226]: MKS socket connecting to 192.168.0.131:8080
2026-10-06 13:48:28,814 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [830]: _send_data: M105  M997
2026-10-06 13:48:30,813 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [830]: _send_data: M105  M997
2026-10-06 13:48:32,813 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [830]: _send_data: M105  M997
2026-10-06 13:48:34,814 - DEBUG - [MainThread] MKSWifiPlugin.MKSOutputDevice.write_socket_data [830]: _send_data: M105  M997```

nicf82, 2026-10-06:
I figured out that at least in macos Sequoia 15.7, you need to go to Settings > Privacy and Security > Developer Tools and add Ultimaker Cura so that it is allowed to open a raw unencrypted socket to the wifi module. Also add Terminal if you want to test the connection using nc/telnet.