PrintMakerLab/mks-wifi-plugin #383: several module

kmz45z, 2023-11-28:
### What is your question?

I have three printers and they are equipped with Wi-Fi modules.
but. Only one printer transmits to wifi successfully.
please Help all your printers work.

Jeredian, 2023-11-28:
Unfortunately our plugin wasn't designed to support multiple connections at once. I'm also not quite sure if Cura provide full support for plugins for it.
To Implement it we need to rework it at all, which may take a lot of time for us ( based on the fact that it's an open source project with 2 active maintainers ). Maybe we'll look at this feature next year. but I can't say for sure.

As a workaround, you can disconnect from one printer and connect to another printer in the list. This will give you the ability to send files to these printers one at a time. However, you will only be able to view statistics for one printer.
Also you may try getting some raspberry pi's and putting Octoprint on them. 

Can you describe your vision for this feature ? What are the main scenarios it should cover for you to be comfortable in it.

kmz45z, 2023-12-02:
I can't think of any scenario right now
Thank you for your response!

closed 2023-12-02 by kmz45z