PrintMakerLab/mks-wifi-plugin #402: every since latest few updates, Cura 5.5.0 constantly disconnects and reconnects from printer. Neptune 2 s. Printer shows steady wifi connection

NXS26, 2024-02-19:
### Plugin Version

latest few versions. 

### Cura Version

5.5.0

### Platform

11

### Printer

Neptune 2 s

### Reproduction steps

uninstalled reinstall from market place, manually install via drag and drop, rolled back to a few previous version of MKS

### Actual results

same issue

### Expected results

cura stays connected

### Checklist of files to include

- [X] Log file

### Additional information & file uploads

there are no log files

Jeredian, 2024-02-19:
Please provide cura.log file

```
How to find Cura log file:
- Windows:
  - %APPDATA%\cura<Cura version>\cura.log
  - C:\Users\\AppData\Roaming\cura<Cura version>\cura.log
- OSX:
  - $USER/Library/Application Support/cura//cura.log
- Linux:
  - $USER/.local/share/cura//cura.log
```

Jeredian, 2024-02-19:
Possible duplication of https://github.com/PrintMakerLab/mks-wifi-plugin/issues/351

Jeredian, 2024-02-19:
Also please try this workaround:
* Close all Cura instances
* Go to the Cura plugins folder
```
- Windows:
  - %APPDATA%\cura\<Cura version>\plugins
  - C:\Users\\AppData\Roaming\cura\<Cura version>\plugins
- OSX:
  - $USER/Library/Application Support/cura/<Cura version>/plugins
- Linux:
  - $USER/.local/share/cura/<Cura version>/plugins
```
* Manually remove **MKSWifiPlugin**
* Run single instances of Cura
* Manually install plugin via drag and drop
* Restart Cura
* Try to connect

Due to hardware limitations, the printer board can only support one connection an a time.
That's why we need to keep only **1 Cura instance** running.
Sometimes when plugin updates Cura does not close previous connections, and therefore it is necessary to restart the application, or sometimes even the system itself.

NXS26, 2024-02-20:
HI, I  have no results. this is the only thing in that folder. and a search never finishes.

NXS26, 2024-02-20:
HI, I don't know why, but my folder structure looks nothing like this.  I can't even find a Cura plugin folder by running a general search on the whole PC. I did try to uninstall through marketplace and reinstall by drag an drop in one instance only running. I generally only have one instance open anyway.

Jeredian, 2024-02-20:
@NXS26 , you can try another way. Inside Cura application, from the application menu in Help -> Show settings folder .
It will open the same folder. Because _AppData_ sometimes hidden on windows os.

[image]

Then you need to send "cura" (aka cura.log) file, it's the single one text document inside that folder :
[image]

And for plugin - it placed in the "plugins" folder

[image]

Hope, it helps.

Could you try to:

1.  Clean up plugin settings (list of IP adresses)
2.  Close Cura
3.  Restart PC
4.  Run Cura 
5.  Set IP address of your printer (in the plugin settings dialog)
6. And click to connect to the printer
7. If the issue still be there, please, send log file ( according to the screenshots above)

I will try to reproduce same thing too, but it's hard to catch.