PrintMakerLab/mks-wifi-plugin #403: Plugin crashes with unassigned variable alert.

gostapps, 2024-02-20:
### Plugin Version

1.4.4

### Cura Version

5.6.0

### Platform

Linux GNU

### Printer

Flying Bear Ghost 6

### Reproduction steps

Plugin generates an error, saying that simage variable was used before assignment. 

### Actual results

Plugin abandones.

### Expected results

Plugin expected to work normally until slice uploads to the printer

### Checklist of files to include

- [ ] Log file

### Additional information & file uploads

Assign variable in MKSPreview.py file:

+++ MKSPreview.py	2024-02-20 20:19:35.684697924 +0100
@@ -114,6 +114,7 @@
 
 def generate_preview(global_container_stack, image):
     screenshot_string = ""
+    simage = 0
     meta_data = global_container_stack.getMetaData()
     Logger.log("d", "Get current preview settings.")
     encoded = False

Jeredian, 2024-02-21:
Thank you for submitting a bug report. 
I will try to reproduce it on macOS and Windows.

In the meantime, could you please provide the cura.log file with this error, it may help in reproducing the problem.
Or just log section related to the issue ( error and 20-30 lines above )

```
How to find Cura log file:
`$CURA_VERSION` is the current version of Cura that's installed.
`$HOME` is your user's home directory, e.g. /home/user.
`$USER` is your user name

- Windows:
  - %APPDATA%\cura\$CURA_VERSION\cura.log
  - C:\Users\$USER\AppData\Roaming\cura\$CURA_VERSION\cura.log
- OSX:
  - $User/Library/Application\ Support/Cura/$CURA_VERSION/cura.log
- Linux:
  - $HOME/.local/share/cura/$CURA_VERSION/cura.log
 - Inside Cura application:
   - from the application menu in Help -> Show settings folder .
   
```

[image]

Jeredian, 2024-02-21:
@gostapps ,   what settings for the previews are you using ?

Jeredian, 2024-02-21:
Reproduced.
Settings:
[image]
Error:
[image]