PrintMakerLab/mks-wifi-plugin #409: Referencing 'simage' before first use.

rburema, 2024-02-21:
### Plugin Version

1.4.4

### Cura Version

5.5

### Platform

Windows 10

### Printer

any

### Reproduction steps

Save.

### Actual results

2024-02-21 23:08:40,137 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file EN3PRO_hires_thing: local variable 'simage' referenced before assignment
2024-02-21 23:08:40,138 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2024-02-21 23:08:40,138 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2024-02-21 23:08:40,139 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2024-02-21 23:08:40,140 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Program Files\UltiMaker Cura 5.5.0\share\uranium\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 139, in requestWrite
2024-02-21 23:08:40,140 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self._performWrite(file_name, selected_type, file_handler, nodes)
2024-02-21 23:08:40,141 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Program Files\UltiMaker Cura 5.5.0\share\uranium\plugins\LocalFileOutputDevice\LocalFileOutputDevice.py", line 160, in _performWrite
2024-02-21 23:08:40,142 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.writeStarted.emit(self)
2024-02-21 23:08:40,142 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2024-02-21 23:08:40,142 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2024-02-21 23:08:40,143 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2024-02-21 23:08:40,144 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2024-02-21 23:08:40,145 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 361, in __performEmit
2024-02-21 23:08:40,145 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     signal.emit(*args, **kwargs)
2024-02-21 23:08:40,146 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2024-02-21 23:08:40,146 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2024-02-21 23:08:40,147 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2024-02-21 23:08:40,147 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2024-02-21 23:08:40,148 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 353, in __performEmit
2024-02-21 23:08:40,148 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     func(*args, **kwargs)
2024-02-21 23:08:40,149 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\<snip>\AppData\Roaming\cura\5.5\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSPreview.py", line 159, in add_preview
2024-02-21 23:08:40,150 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     simage, gimage, screenshot_string = generate_preview(global_container_stack, image)
2024-02-21 23:08:40,151 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\<snip>\AppData\Roaming\cura\5.5\plugins\MKSWifiPlugin\MKSWifiPlugin\MKSPreview.py", line 132, in generate_preview
2024-02-21 23:08:40,151 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     return simage,gimage,screenshot_string
2024-02-21 23:08:40,152 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: UnboundLocalError: local variable 'simage' referenced before assignment

### Expected results

Not throw an error when trying to save.

### Checklist of files to include

- [X] Log file

### Additional information & file uploads

Relevant bit from the log file included in 'actual results'.

Jeredian, 2024-02-22:
@rburema , a pre-release with fixes is available: [1.4.5-rc](https://github.com/PrintMakerLab/mks-wifi-plugin/releases/tag/1.4.5-rc) .

Jeredian, 2024-02-22:
duplicate of #403

Jeredian, 2024-02-22:
@rburema , thank you for submitting bug report. I'll post a new version version to the portal in the next couple days.

closed 2024-02-22 by Jeredian