PrintMakerLab/mks-wifi-plugin #413: Printing does not start after upload

BenTdesign, 2024-06-24:
### Plugin Version

1.4.5

### Cura Version

5.7.1

### Platform

Windows 10

### Printer

Tevo Tarantula

### Reproduction steps

Open cura
Open model
Click "Print over Tevo Tarantula"

### Actual results

File uploads OK.
Print does not start, no preheat.
Printer becomes unresponsive.

Can upload without printing, and select file from printer screen and print OK, but not print from Cura.

### Expected results

Printing!

### Checklist of files to include

- [X] Log file

### Additional information & file uploads

[cura.log](https://github.com/user-attachments/files/15949717/cura.log)

Jeredian, 2024-06-24:
@BenTdesign , what firmware are you using? Does your firmware support M23 and M24 commands?

Could you also try to print a file from the sd-card using the Cura interface. Because in this case the same M23 and M24 will be called, but for an already existing file.
If the printer does not respond, you can call M25 manually. By default, M25 is for pause and M24 for continue, but in some firmware it is the opposite.

BenTdesign, 2024-06-24:
OK, updated firmware to latest version, and it now works, thanks!

Second observation:

Delete files from SD does not work.  Print from SD works OK.  Log file shows M30 command being sent, but file is not deleted.  Tried all options for drive prefix, but no change.  Any ideas?

Jeredian, 2024-06-25:
For my printer `1:/` worked, but it took a while when the file was huge. for default Marlin firmware it should be ""  (empty string). I think if this 2 doesn't work, you can ask on [reddit](https://www.reddit.com/r/tevotarantula/) and we can add it to the dropdown list

However, when sending the command manually (from cura ui, f.ex. "M30 1:/filename.gcode M20") it would never work because the filename will be converted to upper case. - I will create a ticket for this.

Jeredian, 2024-06-25:
@Elkin-Vasily , could there be other drive prefixes?

Elkin-Vasily, 2024-06-25:
> @Elkin-Vasily , could there be other drive prefixes?

Haven't seen those in practice.

BenTdesign, 2024-06-25:
I tried deleting a file with a manual command and that also did not work.  Then made a file with uppercase filename and tried again, and that also didn't work manually or through the more/SD files menu.

Jeredian, 2024-06-26:
Maybe it's a specifics or bug of Tevo Tarantula's firmware. Could you ask on [reddit](https://www.reddit.com/r/tevotarantula/)  about it, and we can keep this thread for future updates.